ArduPilot/node-mavlink #2: Detect the landed state?

codysmallmak, 2021-09-27:
I am trying to figure out how to detect if the drone is taking off, in the middle of flying or landing.  when I read the mavlink documentation it seems like "MAV_LANDED_STATE" is what I want.   I understand that I can send a message or command tot he drone like this 
```
  const message = new common.RequestDataStream();
  message.targetSystem = 1;
  message.targetComponent = 0;
  message.reqStreamId = 0;
  message.reqMessageRate = 1000;
  message.startStop = 0;
  message.paramId = "SYSTEM_TIME";
await send(mavlinkPort, message);
```

but how to I send a message saying send me back the the value of the landed state enum?

padcom, 2021-09-27:
Well, looking at the mappings, there are 3 options:

- HighLatency (which is deprecated)
- ExtendedSysState (I guess this is the one you'd like to use)
- AutopilotStateForGimbalDevice (you probably don't want this one)

All the above provide the field `landedState` of type MavLandedState. You
can either request that the status is delivered in intervals
(MavCmd.SET_MESSAGE_INTERVAL) or a one-off (MavCmd.REQUEST_MESSAGE).
So the code would look something like this:

```
const message = new common.CommandLong();
message.targetSystem = 1;
message.targetComponent = 0;
message.command = MavCmd.REQUEST_MESSAGE
message.paramId1 = ExtendedSysState.MSG_ID;
await send(mavlinkPort, message);
``` or if you'd like to get the status periodically then ``` const message
= new common.CommandLong()
message.targetSystem = 1
message.targetComponent = 0
message.command = MavCmd.SET_MESSAGE_INTERVAL
message.paramId1 = ExtendedSysState.MSG_ID
message.paramId2 = 1000 // period of how often should the message be
delivered (see
https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL)
await send(mavlinkPort, message)
``` Check out also
https://discuss.px4.io/t/reading-of-mav-landed-state-from-mavlink/9962
which has some C examples on how to request status. Also, looks like the
*REQUEST_DATA_STREAM* command is deprecated in favor of
*SET_MESSAGE_INTERVAL* so you should be better of using the current version
instead of the deprecated one.

Hope this helps.

Cheers,
Matthias.

pon., 27 wrz 2021 o 19:33 codysmallmak ***@***.***>
napisał(a):

> I am trying to figure out how to detect if the drone is taking off, in the
> middle of flying or landing. when I read the mavlink documentation it seems
> like "MAV_LANDED_STATE" is what I want. I understand that I can send a
> message or command tot he drone like this
>
>   const message = new common.RequestDataStream();
>   message.targetSystem = 1;
>   message.targetComponent = 0;
>   message.reqStreamId = 0;
>   message.reqMessageRate = 1000;
>   message.startStop = 0;
>   message.paramId = "SYSTEM_TIME";
> await send(mavlinkPort, message);
>
> but how to I send a message saying send me back the the value of the
> landed state enum?
>
> —
> You are receiving this because you are subscribed to this thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/padcom/node-mavlink/issues/2>, or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/AAEHICSYF7I7A7RBVULZACTUECTGJANCNFSM5E3CSKVQ>
> .
> Triage notifications on the go with GitHub Mobile for iOS
> <https://apps.apple.com/app/apple-store/id1477376905?ct=notification-email&mt=8&pt=524675>
> or Android
> <https://play.google.com/store/apps/details?id=com.github.android&referrer=utm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>.
>
>

padcom, 2021-09-27:
Well, looking at the mappings, there are 3 options:

- `HighLatency` (which is deprecated)
- `ExtendedSysState` (I guess this is the one you'd like to use)
- `AutopilotStateForGimbalDevice` (you probably don't want this one)

All the above provide the field `landedState` of type `MavLandedState`. You
can either request that the status is delivered in intervals
(`MavCmd.SET_MESSAGE_INTERVAL`) or a one-off (`MavCmd.REQUEST_MESSAGE`).
So the code would look something like this:

```
const message = new common.CommandLong();
message.targetSystem = 1;
message.targetComponent = 0;
message.command = MavCmd.REQUEST_MESSAGE
message.paramId1 = ExtendedSysState.MSG_ID;
await send(mavlinkPort, message);
```

or if you'd like to get the status periodically then 

``` 
const message = new common.CommandLong()
message.targetSystem = 1
message.targetComponent = 0
message.command = MavCmd.SET_MESSAGE_INTERVAL
message.paramId1 = ExtendedSysState.MSG_ID
message.paramId2 = 1000 // period of how often should the message be delivered (see https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL)
await send(mavlinkPort, message)
```
Check out also
https://discuss.px4.io/t/reading-of-mav-landed-state-from-mavlink/9962
which has some C examples on how to request status. Also, looks like the
`REQUEST_DATA_STREAM` command is deprecated in favor of
`SET_MESSAGE_INTERVAL` so you should be better off using the current version
instead of the deprecated one.

Hope this helps.

padcom, 2021-09-27:
Oh, and I noticed you tried using the message that is sent from the autopilot down to your app as the constructor. This is not how it works. The construction of the message is done by the autopilot and then the message is serialized and sent back to you. All commands are requested by either `CommandInt` or (more popular) `CommandLong` messages (as shown in the example above).

codysmallmak, 2021-09-27:
> Oh, and I noticed you tried using the message that is sent from the autopilot down to your app as the constructor. This is not how it works. The construction of the message is done by the autopilot and then the message is serialized and sent back to you. All status commands are requested by either `CommandInt` or (more popular) `CommandLong` messages (as shown in the example above) where the `param1` field is set to the message you are requesting. In the case of periodic updates sent by the autopilot you can set the `paramId2` field to either `0` (default time interval), -1 to stop sending messages or a number of milliseconds of delay between messages are being sent.

That helps a ton thank you!  

so a believe I have update the code to where it is sending the status request over properly as I am getting a command ack response when I run the program but know do I access the response?  

I send the requested over with the below code
```
const mavlinkPort = new SerialPort("/dev/ttyACM0");

// constructing a reader that will emit each packet separately
const REGISTRY = {
  ...minimal.REGISTRY,
  ...common.REGISTRY,
  ...ardupilotmega.REGISTRY,
  ...uavionix.REGISTRY,
  ...icarous.REGISTRY,
};

  const message = new common.CommandLong();
  message.targetSystem = 1;
  message.targetComponent = 0;
  message.command = common.MavCmd.SET_MESSAGE_INTERVAL;
  message.paramId1 = common.ExtendedSysState.MSG_ID;
  message.paramId2 = 1000;
  await send(mavlinkPort, message);
  console.log("message SENT", message);
```

and try reading the response with this 
```
const reader = mavlinkPort
  .pipe(new MavLinkPacketSplitter())
  .pipe(new MavLinkPacketParser());

reader.on("data", (packet) => {
  //console.log(packet.debug());
  const clazz = REGISTRY[packet.header.msgid];
  if (clazz) {
    const data = packet.protocol.data(packet.payload, clazz);
    if (data.constructor.name != "Heartbeat") {
      console.log("Received packet:", data);
    }
  }
});
``` 

but I don't seem to be getting a repeated output of messages every 1000ms

padcom, 2021-09-27:
Honestly I don't know. It should print out all the packets that are coming in from the autopilot. Unfortunately I don't have a system that I could test it with now :( so I won't be able to be of more help to you until I get one (which is not coming soon since I have moved over to INAV lately)

padcom, 2021-09-27:
The only thing that comes to mind is that you're sending the message before the actual connection is made and the packet is lost before being transferred over the wire.

padcom, 2021-09-28:
Just out of curiosity: I can see the `REGISTRY` being built and right after that I can see the message is being sent. It seems like you're using parts of the examples. This means that the code sending anything is executed before you are connected.

That part:
```
port.on('open', async () => {
  // the port is open - we're ready to send data
  await send(port, msg, new MavLinkProtocolV2())
})
```

is there for a reason (https://github.com/padcom/node-mavlink#sending-messages)

closed 2021-09-28 by padcom