ArduPilot/node-mavlink #3: Dropping MavlinkV2 data

gribbet, 2021-09-29:
I have a particular V2 packet that happens to have the Mavlink V1 STX somewhere in it.
The current behavior of the MavLinkPacketSplitter is to assume a V1 packet if the V1 STX exists anywhere within the packet so the data is dropped as invalid. Even though the first byte in the buffer is the V2 STX.

https://github.com/padcom/node-mavlink/blob/917762a9026e910f9a846e612b7a1198dae356b8/lib/mavlink.ts#L458-L468

Perhaps the MavLinkSplitter should save the last used protocol version and try that one first?

padcom, 2021-09-29:
Well, it's not quite as simple as that. First of all the splitter searches for the start of a packet (STX) but the next packet is then read after skipping as many bytes as the recognized protocol defines. This means that any in-the-middle values that might have the same value as V1 STX or V2 STX will not be taken into account because they will be consumed from the input stream and only after that the next start of text will be analyzed. 

If you have an example that could visualize your problem, that would help a lot. Maybe there is a bug in the splitter but the general idea remains the same: find STX, read as many bytes as the packet contains, read the next STX.

padcom, 2021-09-29:
Could you record the stream of events that you're getting so that I can reproduce the case locally?

gribbet, 2021-09-29:
Here's an example MavlinkV2 packet that cannot be handled by MavLinkSplitter in base64: ```/SUAAM0BAUkAAAAAAAAAAAAAAAAAAAAAwH/+V4wWULb8tuXyR0JbABAAAQIGAAEyXQ==```

For example:
```
import { MavLinkPacketParser, MavLinkPacketSplitter } from "node-mavlink";
import { Readable } from "stream";

const readable = new Readable();
readable._read = () => [];
for (let i = 0; i < 100; i++) {
  readable.push(
    Buffer.from(
      "/SUAAM0BAUkAAAAAAAAAAAAAAAAAAAAAwH/+V4wWULb8tuXyR0JbABAAAQIGAAEyXQ==",
      "base64"
    )
  );
}
const reader = readable
  .pipe(new MavLinkPacketSplitter({}, true))
  .pipe(new MavLinkPacketParser());

reader.on("data", _ => console.log("Packet", _));
```

results in many
```
mavlink.js:433 CRC error; expected 61588 (0xf094) got 0 (0x0000) ; msgid: 182 , magic: 229
utils.js:38 fe  57  8c  16  50  b6  fc  b6  e5  f2  47  42  5b  00  10  00  01  02  06  00  01  32  5d  fd  25  00  00  cd 
utils.js:38 01  01  49  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  c0  7f  fe  57  8c  16  50  b6  fc 
utils.js:38 b6  e5  f2  47  42  5b  00  10  00  01  02  06  00  01  32  5d  fd  25  00  00  cd  01  01  49  00  00  00  00 
utils.js:43 00  00  00  00  00  00  00  00  00  00  00 
```

If I make a change in MavLinkSplitter to try V2 first then the parsing succeeds.

padcom, 2021-09-29:
@gribbet That is something I can work with. Gimme some time to look into it.

gribbet, 2021-09-29:
Thank you @padcom 
Please let me know if I can assist in any way.

padcom, 2021-09-29:
@gribbet OK, so I have analyzed the problem. Regardless of which protocol is being tried first, there always will be a combination of bytes inside a packet that will throw this algorithm off balance. Just like in the example you provided, if there is the 0xFE inside, it will still fail to parse the packet. Such is the nature of the adaptive splitter, which tries to be an all-in-one solution. It tries parsing packets until it finds the first one that is correct (which the 100 buffers you're creating will never be) and once a correct packet is retrieved the rest of the packets will be processed properly because of the payload length information in the packet.

A similar situation happens when you have a value like 0xFD or 0xFE inside the packet and the stream starts analyzing it from the data byte instead of the STX. You need to skip that byte and try again, basically dropping the incomplete frame until a proper packet is found. If a packet is "proper" its CRC will match the expected value, and we know for sure we have a proper packet.

The buffer you provided is a constant stream of invalid V1/V2 packets in this scenario.

Try capturing a bit more packets that have the same protocol, and you'll notice that once the first proper packet is found, the rest will be properly recognized. 

On the other note, would you tell me how did you come by the data? Is it being sent from some system that you have? Which version is it?

padcom, 2021-09-29:
@gribbet Also note that the protocol switch happens only once. The procedure is described here: https://mavlink.io/en/guide/mavlink_version.html#version_handshaking. Depending on the device and capabilities you'd like to explore, you can stick to v1 and it will be good enough (see example from the node-mavlink readme). This means that once you have negotiated protocol upgrade from v1 to v2, all the other received packets will be v2, not a mixture of v1 and v2. In this case, the adaptive nature will kick in and all all the packets will be recognized properly.

The only solution that would give you 100% chance of decoding a mixture of V1/V2 packets it to manually switch the protocol in splitter, but I don't think that makes sense.

gribbet, 2021-09-29:
> On the other note, would you tell me how did you come by the data? Is it being sent from some system that you have? Which version is it?

I am streaming Mavlink V2 data from a simulated Pixhawk and am seeing significant packet loss. Further analysis revealed that one particular packet (a mission item packet) would never be parsed even after re-requesting it many times.

gribbet, 2021-09-29:
> Try capturing a bit more packets that have the same protocol, and you'll notice that once the first proper packet is found, the rest will be properly recognized.

This is not what I'm seeing. Many successful V2 packets are parsed and then eventually it hits one with a V1 STX and parsing fails and packets are dropped

padcom, 2021-09-29:
@gribbet Can you get me a longer buffer data? Maybe via a gist... let's say 100kb or more

gribbet, 2021-09-29:
[mavlink.bin.zip](https://github.com/padcom/node-mavlink/files/7254899/mavlink.bin.zip)

This is raw MavlinkV2 data. No CRC errors. 3412 packets. MavLinkSplitter drops 99% of them.

padcom, 2021-09-29:
Thanks! Will look at it. That shouldn't happen that we're getting 99% dropped packets in a proper stream of bytes

padcom, 2021-09-30:
So... here's what happens:
- the stream contains a packet that starts with the STX being in the middle of the message
- it assumes it is the start of the message and reads the length of it
- after checking CRC and finding out it is wrong, what it does is it skips the entire length of the message thus most probably landing in the middle of another message

The solution, which I'll try to implement Today, is to skip only 1 byte after the STX of the packet that has been deemed bad through CRC calculation. So instead of skipping over 20k of bytes and missing the next packages it will try harder to find the next STX. Sure, there will be an wrong packet in the beginning (because it is wrong as it is the data not the actual STX) but starting from the next packet it should all be OK.

padcom, 2021-09-30:
@gribbet Check out version 1.0.4. It should work with the above fix applied. At least the whole file is parsed with 0 bad packets. Check out examples/send-receive-file.ts.

padcom, 2021-09-30:
@gribbet I've also done some cleanup of the code as it was messy enough to be hard to read and change. Version 1.0.5 is released.

gribbet, 2021-09-30:
@padcom I'm seeing much better results with 1.0.6. You're awesome! Thank you.

closed 2021-09-30 by gribbet