ArduPilot/node-mavlink #5: error: "value of offset is outofrange"

mnumanuyar, 2022-03-18:
Hello,
I am using windows and connecting to a cubeblack pix4 via serial usb.
I am getting this error, I assume it is some weird conversion thing (it is actually 0 but instead thinks it is 256) 

also when I printed "clazz" i get 

```
[class Odometry extends MavLinkData] {
  MSG_ID: 331,
  MSG_NAME: 'ODOMETRY',
  PAYLOAD_LENGTH: 232,
  MAGIC_NUMBER: 91,
  FIELDS: [
    MavLinkPacketField {
      name: 'timeUsec',
      offset: 0,
      type: 'uint64_t',
      length: 0,
      extension: false,
      size: 8,
      units: 'us'
    },
```

I am not sure why timeUsec has length 0, ofset 0.

the error I get:

RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 251. Received 256
    at new NodeError (node:internal/errors:371:5)
    at boundsError (node:internal/buffer:86:9)
    at Buffer.readFloatForwards [as readFloatLE] (node:internal/buffer:532:5)
    at float[] (D:\GitRepos\yki\node_modules\node-mavlink\dist\lib\serialization.js:163:32)
    at D:\GitRepos\yki\node_modules\node-mavlink\dist\lib\mavlink.js:51:36
    at Array.forEach (<anonymous>)
    at MavLinkProtocolV2.data (D:\GitRepos\yki\node_modules\node-mavlink\dist\lib\mavlink.js:46:22)
    at MavLinkPacketParser.<anonymous> (D:\GitRepos\yki\libs\util\serial-standalone.js:34:38)
    at MavLinkPacketParser.emit (node:events:390:28)
    at addChunk (node:internal/streams/readable:315:12) {
  code: 'ERR_OUT_OF_RANGE'
}

mnumanuyar, 2022-03-18:
on second look I realised i didnt get errors in some other messages with fields length 0 offset 0

mnumanuyar, 2022-03-18:
when I make fallowing change in serialization.js ~line 160

```
    'float[]': (buffer, offset, length) => {
        const result = new Array(length);
        for (let i = 0; i < length; i++){
            console.log(offset,offset + i * 8,i )
            result[i] = buffer.readFloatLE(offset + i * 8);
        }
        return result;
    }
```
it prnts "144 256 14" before error

padcom, 2022-03-18:
It's a bit hard to understand the problem without something to reproduce it. If you could record the faulty transmission to a binary file and attach it to the ticket I could better understand what the problem actually is.

mnumanuyar, 2022-03-18:
ok thanks I will try it. I am new to JS, do you have some sample code for it?

right now my code looks like this: 

```
const { SerialPort } = require('serialport');
const { MavLinkPacketSplitter, MavLinkPacketParser, minimal, common, ardupilotmega, uavionix, icarous,
    asluav, development, matrixpilot, paparazzi, ualberta,send,MavLinkProtocolV2 } = require('node-mavlink')

// substitute /dev/ttyACM0 with your serial port!
const recievePort = "COM8";
const baudRate = 115200;
const port = this.socket = new SerialPort({
    path: recievePort,
    baudRate: baudRate
});
const REGISTRY = {
    ...minimal.REGISTRY,
    ...common.REGISTRY,
    ...ardupilotmega.REGISTRY,
    ...uavionix.REGISTRY,
    ...icarous.REGISTRY,
    ...asluav.REGISTRY,
    ...development.REGISTRY,
    ...matrixpilot.REGISTRY,
    ...paparazzi.REGISTRY,
    ...ualberta.REGISTRY,
}
// constructing a reader that will emit each packet separately
const reader = port
    .pipe(new MavLinkPacketSplitter())
    .pipe(new MavLinkPacketParser())

reader.on('data', packet => {
    const clazz = REGISTRY[packet.header.msgid]
    if (clazz) {
        console.log( clazz.MSG_NAME)
        console.log( clazz)
        const data = packet.protocol.data(packet.payload, clazz)
    }
})

const msg = new common.CommandInt()
msg.command = common.MavCmd.REQUEST_PROTOCOL_VERSION
msg.param1 = 1

port.on('open', async () => {
  // the port is open - we're ready to send data
  await send(port, msg, new MavLinkProtocolV2())
})

```

mnumanuyar, 2022-03-18:
[secret.txt](https://github.com/ArduPilot/node-mavlink/files/8303393/secret.txt)

is this usefull ?

padcom, 2022-03-18:
I will take a look at it Today in the evening. Stay tuned.

mnumanuyar, 2022-03-18:
I have checked, and as ı suspected, the error is not in that file.

however i manage to get error from this file
[output.txt](https://github.com/ArduPilot/node-mavlink/files/8304263/output.txt)

using this code:

```javascript
const { SerialPort } = require('serialport');
const { MavLinkPacketSplitter, MavLinkPacketParser, minimal, common, ardupilotmega, uavionix, icarous,
    asluav, development, matrixpilot, paparazzi, ualberta, send, MavLinkProtocolV2 } = require('node-mavlink')
const { fs, createReadStream } = require('fs');

const file = createReadStream('./output.txt')

const splitter = new MavLinkPacketSplitter({}, true)

const fileReader = file
    .pipe(splitter)
    .pipe(new MavLinkPacketParser())
//const Readline = require("@serialport/parser-readline");

const REGISTRY = {
    ...minimal.REGISTRY,
    ...common.REGISTRY,
    ...ardupilotmega.REGISTRY,
    ...uavionix.REGISTRY,
    ...icarous.REGISTRY,
    ...asluav.REGISTRY,
    ...development.REGISTRY,
    ...matrixpilot.REGISTRY,
    ...paparazzi.REGISTRY,
    ...ualberta.REGISTRY,
}

fileReader.on('data', packet => {
    const clazz = REGISTRY[packet.header.msgid]
    if (clazz) {
        console.log(clazz.MSG_NAME)
        console.log(clazz)
        const data = packet.protocol.data(packet.payload, clazz)
    }
})

```

mnumanuyar, 2022-03-18:
Thank you again for fast responses :D

padcom, 2022-03-18:
Version 1.0.9 should have the problem fixed. It was a typo coming from copy/pasting code (shame on me :D), where floats have been designated to have 8 bytes instead of 4. Therefore, `i * 4 + offset` was beyond the buffer, hence the error.

Please let me know if it fixed your problem (the file did parse without issues, I checked!) and if so, please close the ticket.

mnumanuyar, 2022-03-21:
Thank you so much :D, version 1.0.9 worked !

closed 2022-03-21 by mnumanuyar