ArduPilot/node-mavlink #6: Issue with serializer uint8_t[]

thiensu1122, 2022-03-29:
Hi team,

I've got into a problem when creating GpsInjectData - this is my log of the message
 {
  targetSystem: 1,
  targetComponent: 0,
  len: 110,
  data: <Buffer a0 a1 03 72 e5 01 e4 08 9b 0d a9 bd e0 03 e8 00 00 1c 05 0d c0 2a 41 82 1f 81 a4 63 b9 8c c1 98 f4 56 9e 92 00 00 44 c6 e0 00 00 00 00 00 07 00 00 02 ... 60 more bytes>
}

Yet when I try to send this message I got the error:

(node:1149) Error: Unknown field type uint8_t[]: serializer not found
    at /home/pi/companion/companion-nodejs/node_modules/node-mavlink/dist/lib/mavlink.js:161:23
    at Array.forEach (<anonymous>)
    at MavLinkProtocolV2.serialize (/home/pi/companion/companion-nodejs/node_modules/node-mavlink/dist/lib/mavlink.js:158:27)
    at /home/pi/companion/companion-nodejs/node_modules/node-mavlink/dist/lib/mavlink.js:649:37
    at new Promise (<anonymous>)
    at /home/pi/companion/companion-nodejs/node_modules/node-mavlink/dist/lib/mavlink.js:648:16
    at Generator.next (<anonymous>)
    at /home/pi/companion/companion-nodejs/node_modules/node-mavlink/dist/lib/mavlink.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/home/pi/companion/companion-nodejs/node_modules/node-mavlink/dist/lib/mavlink.js:4:12)

Can you tell me how to fix this?

padcom, 2022-03-29:
Looks like somehow the type is wrong. It should be `uint8[]` - without that `_t` in the name. Let me check the definition and maybe I can do something about it.

padcom, 2022-03-29:
Ok, looks like it was a major issue with serialization of arrays! Good catch!

Please check out version `1.0.10-beta.1` - it should work.

closed 2022-03-29 by padcom