ArduPilot/node-mavlink #13: Serialization error regarding uint64_t type fields

SKYnSPACE, 2022-06-22:
Whenever I try to serialize a payload containing uint64_t field, such as SystemTime(#2) in the common dialect, the following error arises:

node:internal/buffer:581
  let lo = Number(value & 0xffffffffn);
                        ^
TypeError: Cannot mix BigInt and other types, use explicit conversions
    at writeBigU_Int64LE (node:internal/buffer:581:25)
    at Buffer.writeBigUInt64LE (node:internal/buffer:601:10)

Are there any other protocols to follow to handle the types in 64 bits?
Thank you :)

padcom, 2022-06-22:
64-bit numbers in Node.js are represented with BigInt. Check out the docs at https://developer.mozilla.org/pl/docs/Web/JavaScript/Reference/Global_Objects/BigInt to learn how to deal with those values.

closed 2022-06-22 by padcom

padcom, 2022-06-22:
It essentially means you need to use the `n` suffix :)

[image]