ArduPilot/node-mavlink #14: Wrong version of mappings has resolved under installation

n2j7, 2022-07-02:
There are `"mavlink-mappings": "^1.0.9-20220424"` in dependecies of this module at `package.json`
But when npm works it resolve this into `1.0.9-20220424-2` with **-2** in the end because of `^` in the beginning of version in package.json
And there is a problem due to that version doesn't export magic numbers

Just compare them:
https://github.com/ArduPilot/node-mavlink-mappings/blob/1.0.9-20220424/index.ts#L4
https://github.com/ArduPilot/node-mavlink-mappings/blob/1.0.9-20220424-2/index.ts

I have an error like this
```
 UnhandledPromiseRejectionWarning: TypeError: Cannot read property '30' of undefined
    at MavLinkPacketSplitter.validatePacket (/home/pi/copilot/node_modules/node-mavlink/dist/lib/mavlink.js:1:10368)
```

Everyone can fix that with `npm i mavlink-mappings@1.0.9-20220424` after installing node-mavlink
I'm using `raspberry pi zero w+` under  Raspbian 11 with node `v12.22.12` and npm `7.5.2` installed with package manager

P.S.: This is the solution for get things working right now and require other repository fix

padcom, 2022-07-02:
Could you try it out with version `1.0.10-20220702-2`? The one without -2 has still the minificator enabled, the one with -2 has it disabled. If you could both versions and share the results here, that'd be awesome.

n2j7, 2022-07-05:
It will not work with `1.0.10-20220702-2` and it does not work  for the same reason :(
We haven't this export `export * from './lib/magic-numbers'` from `node-mavlink-mappings`
So, in basic example
```
const port = new SerialPort({ path: '/dev/ttyACM0', baudRate: 115200 })

const reader = port
  .pipe(new MavLinkPacketSplitter()) // <-- error here

reader.on('data', packet => {
  console.log(packet)
})
```

We have an error due to when we split our buffer we call `this.validatePacket(buffer, Protocol)`
And this package validation procedure is based on `MSG_ID_MAGIC_NUMBER`
https://github.com/ArduPilot/node-mavlink/blob/333ef9fa154e7e2367c41c287e4bd9e3019b795d/lib/mavlink.ts#L610

Which comes from `import { MSG_ID_MAGIC_NUMBER } from 'mavlink-mappings'`
https://github.com/ArduPilot/node-mavlink/blob/333ef9fa154e7e2367c41c287e4bd9e3019b795d/lib/mavlink.ts#L4

But `node-mavlink-mappings` doesn't export it since this commit
https://github.com/ArduPilot/node-mavlink-mappings/commit/ec3ac84d487dbc903d47495bb887ecbefca27e67

So, for anyone, even simple example is totally broken :(

And again about versions.
`node-mavlink` has this line
https://github.com/ArduPilot/node-mavlink/blob/master/package.json#L25

when `npm` resolve this line it use [this methodology called semver](https://github.com/npm/node-semver#caret-ranges-123-025-004)

So when you wrote something like `B: ^1.2.3` in your dependencies in project A and have published only package B with version `1.2.3` everything if fine. When you publish package B with version `1.3.0` you will not have any problem at all, so everything is fine. But when you publish package B with version `1.2.3any_suffix` or `1.2.4` due to semver rules your main project will get this new version.

This thing happened with `^1.0.9-20220424` and `^1.0.9-20220424-2`

P.S.: sorry if you know all about semver, I'm not trying to offend you, I'm just explain what I mean
P.S.S.: sorry for bad english and don't focus on it (:
P.S.S.S: thank your for awesome and useful package!!!

padcom, 2022-07-06:
Please check `mavlink-mappings@1.0.11-20220706`. I've re-introduced the generation of `magic-numbers.ts`. It was done by mistake during a major cleanup.

padcom, 2022-07-06:
@n2j7 I am not offended. I am sorry that such a mistake on my end caused so much trouble to you and others.

n2j7, 2022-07-06:
I can confirm that it works with `mavlink-mappings@1.0.11-20220706` :blush: 
Thank you! :hugs:

closed 2022-07-06 by padcom