ArduPilot/node-mavlink #15: Trying to run basic example

CarbonMan, 2022-07-03:
Sorry this is going to be a real basic question.
I am just trying to connect to the serial port where my pixhawk is connected. Running on Windows
`
import { SerialPort } from 'serialport'
import { MavLinkPacketSplitter, MavLinkPacketParser } from 'node-mavlink'
import { minimal, common, ardupilotmega, waitFor, send } from 'node-mavlink'

const REGISTRY = {
  ...minimal.REGISTRY,
  ...common.REGISTRY,
  ...ardupilotmega.REGISTRY,
}

async function main() {
  const port = new SerialPort({ path: 'COM4', baudRate: 115200 })

  const reader = port
    .pipe(new MavLinkPacketSplitter())
    .pipe(new MavLinkPacketParser())

  let online = false

  reader.on('data', (packet) => {
    online = true
    const clazz = REGISTRY[packet.header.msgid]
    if (clazz) {
      const data = packet.protocol.data(packet.payload, clazz)
      console.log('>', data)
    } else {
      console.log('!', packet.debug())
    }
  })

  await waitFor(() => online)

  const message = new common.ParamRequestList()
  message.targetSystem = 1
  message.targetComponent = 1

  await send(port, message)
}

main()`

When I run it I get the following output:

`node send-receive2.mjs

(node:10020) UnhandledPromiseRejectionWarning: TypeError: Cannot read property '2' of undefined
    at MavLinkPacketSplitter.validatePacket (D:\learning\MavLink\node_modules\node-mavlink\dist\lib\mavlink.js:1:10368)
    at MavLinkPacketSplitter._transform (D:\learning\MavLink\node_modules\node-mavlink\dist\lib\mavlink.js:1:9389)
    at MavLinkPacketSplitter.Transform._read (internal/streams/transform.js:205:10)
    at MavLinkPacketSplitter.Transform._write (internal/streams/transform.js:193:12)
    at writeOrBuffer (internal/streams/writable.js:358:12)
    at MavLinkPacketSplitter.Writable.write (internal/streams/writable.js:303:10)
    at SerialPort.ondata (internal/streams/readable.js:726:22)
    at SerialPort.emit (events.js:400:28)
    at addChunk (internal/streams/readable.js:290:12)
    at readableAddChunk (internal/streams/readable.js:265:9)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:10020) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:10020) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:10020) UnhandledPromiseRejectionWarning: Timeout
(node:10020) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)`

If I connect to the port just using serialPort I can see that the port is receiving data. What am I doing wrong?

padcom, 2022-07-03:
@CarbonMan I don't have PixHawk hardware to test it with. Could you records the packets coming from your controller to a bin file so that I can reproduce the issue?

whopis, 2022-07-03:
I am having a similar thing happen.  I always get a message saying:

TypeError: Cannot read properties of undefined (reading '165')

or something similar.  The number show (165) above, changes and appears to be the message ID of the MAVLink message that was received.

whopis, 2022-07-03:
@CarbonMan 
I'm not sure if this is the same error you are having, but I was able to resolve mine with the info here:

https://github.com/ArduPilot/node-mavlink/issues/14

CarbonMan, 2022-07-03:
@whopis yes that was it. Thank you.

closed 2022-07-03 by CarbonMan

padcom, 2022-07-05:
I'll take a closer look at it next week, so don't hold your breath until then.