ArduPilot/node-mavlink #18: Send function gives back 41

Simo524, 2022-12-14:
Hello, I'm sending a Mavlink message and everytime i try to send it, it gives back "41"
`const msg = new common.CommandInt();

  msg.command = common.MavCmd.DO_MOUNT_CONTROL;
  msg.param1 = req.body.pitch;
  msg.param2 = req.body.roll;
  msg.param3 = req.body.yaw;
  msg.param7 = 2;
  
  const data = await send(port, msg, new MavLinkProtocolV2());
  console.log(data);`

I really don't know what to do with it, since it is not doing what it should

padcom, 2022-12-15:
@Simo524 Can you give me more information about your setup? Controller, AP version.. anything to pinpoint the problem

padcom, 2022-12-15:
Besides the setup "41" seems to the the correct result. You're sending a command that resulted in transferring 41 bytes over to the aircraft. 

https://github.com/ArduPilot/node-mavlink/blob/master/lib/mavlink.ts#L773

The return value is the number of bytes that have been sent.

Simo524, 2022-12-17:
> @Simo524 Can you give me more information about your setup? Controller, AP version.. anything to pinpoint the problem

@padcom sorry for the late response, i didn't receive any email, i have a pixhawk cube wired with a usb cable and a gremsy S1v3 wired to the pixhawk telem 2 interface, i'm trying to make it change position as i want (pitch, yaw, roll), but it won't move

Simo524, 2022-12-17:
@padcom i also tried using MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW mavlink message with the udid but it also didn't work

padcom, 2022-12-17:
Let's get something straight. The issue is about the `send` function giving back `41`. That's not an issue - that is expected behavior. It gives back as many bytes as were sent, so 41 in this case is correct.

If you have a problem with how your flight controller behaves, I think you should first start with something that you already know or something that has a user interface where you can fiddle around with it - not by starting with a library.

I don't know about the gimbal commands, but you should probably check the response you get from your flight controller to understand what's going on.

closed 2022-12-17 by padcom