ArduPilot/node-mavlink #28: Issues with ArduPilot MAVLink message definitions

gribbet, 2023-06-19:
I've been using this library with the ArduPilot fork of the MAVLink message definitions (https://github.com/ArduPilot/mavlink). To accomplish this I've forked the https://github.com/ArduPilot/node-mavlink-mappings package and updated the message definitions references in package.json to get the desired code generation. This has been working fine until somewhat recently. I'm now running into issues with code generation on the forked message definitions. 

Here's a sample of some of the code generation issues I'm seeing: 
```
lib/ardupilotmega.ts:5136:11 - error TS2304: Cannot find name 'MavResult'.

5136   result: MavResult
               ~~~~~~~~~

lib/ardupilotmega.ts:5136:11 - error TS4031: Public property 'result' of exported class has or is using private name 'MavResult'.

5136   result: MavResult
               ~~~~~~~~~

lib/storm32.ts:758:17 - error TS2552: Cannot find name 'GimbalDeviceErrorFlags'. Did you mean 'GimbalDeviceFlags'?

758   failureFlags: GimbalDeviceErrorFlags
                    ~~~~~~~~~~~~~~~~~~~~~~

lib/storm32.ts:758:17 - error TS4031: Public property 'failureFlags' of exported class has or is using private name 'GimbalDeviceErrorFlags'.

758   failureFlags: GimbalDeviceErrorFlags
```

It looks like that first issue is caused by the fact that ardupilotmega.xml uses MAV_RESULT which is referenced in the include common.xml. It seems like the code generator doesn't handle the include as expected. 

I'd be happy to work on changes necessary to support this use-case if I were given some guidance. Please advise.

padcom, 2023-06-20:
There are templates for imports and if there's something used that is not from imports from that template then you'll run into the kind of trouble you're in.

I'll update the imports in a moment. Please hold.

padcom, 2023-06-20:
@gribbet I've released mavlink-mappings 1.0.19-20230620-0 that use the mavlink-mappings-gen 0.0.9 which does one thing differently:
- for `enum` fields it gets the value of the first item

That was kind of an issue already. Try now, with the latest 1.0.19-20230620-0 and 0.0.9. Close the issue if it fixes it, create a repo with your example otherwise.

padcom, 2023-06-20:
Something must be off on your end. I can build it no problem:

```
padcom@padcom:~/projects/test$ git clone git@github.com:gribbet/node-mavlink-mappings.git
Cloning into 'node-mavlink-mappings'...
remote: Enumerating objects: 521, done.
remote: Counting objects: 100% (165/165), done.
remote: Compressing objects: 100% (43/43), done.
remote: Total 521 (delta 138), reused 134 (delta 122), pack-reused 356
Receiving objects: 100% (521/521), 357.99 KiB | 1.08 MiB/s, done.
Resolving deltas: 100% (371/371), done.

padcom@padcom:~/projects/test/node-mavlink-mappings [master]$ npm i

added 119 packages, and audited 120 packages in 2s

35 packages are looking for funding
  run `npm fund` for details

3 vulnerabilities (2 moderate, 1 high)

To address issues that do not require attention, run:
  npm audit fix

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.
padcom@padcom:~/projects/test/node-mavlink-mappings [master]$ npm run regenerate

> mavlink-mappings@1.0.19-20230620-0 regenerate
> npm run download && npm run code-generator && npm run transpile

> mavlink-mappings@1.0.19-20230620-0 download
> ts-node download.ts

Downloading minimal.xml..........done.
Downloading common.xml..........done.
Downloading ardupilotmega.xml....done.
Downloading uavionix.xml...done.
Downloading icarous.xml...done.
Downloading asluav.xml....done.
Downloading development.xml....done.
Downloading ualberta.xml...done.
Downloading storm32.xml...done.

> mavlink-mappings@1.0.19-20230620-0 code-generator
> ts-node code-generator.ts

Generating code for minimal...done
Generating code for common...done
Generating code for ardupilotmega...done
Generating code for uavionix...done
Generating code for icarous...done
Generating code for asluav...done
Generating code for development...done
Generating code for ualberta...done
Generating code for storm32...done

> mavlink-mappings@1.0.19-20230620-0 transpile
> tsc

padcom@padcom:~/projects/test/node-mavlink-mappings [master]$ 
```

No errors, everything works as expected.

gribbet, 2023-06-20:
> 

Sorry, use the branch [gribbet/ardupilot-mavlink](https://github.com/gribbet/node-mavlink-mappings/tree/gribbet/ardupilot-mavlink)

padcom, 2023-06-20:
OK, I understand what the problem is, and I can do little about it since I don't support that fork of mavlink definitions.

However, in your case, since you're already modifying the source of XML definitions you'll need to adjust the required imports to match whatever is needed. To do that, in those generated files that have issues, in lib/filename.imports.ts add the missing imports and generate the code again with npm run code-generator.

I'll close this issue now since it is about mappings for something else than the officially supported ones.

closed 2023-06-20 by padcom

padcom, 2023-06-20:
I'd like to recognize @gribbet for realizing the potential that configuration done in `package.json` gives. Once the imports definitions will be updated, this is an absolutely valid use case which should be described in the docs.

Thank you!