ArduPilot/node-mavlink #30: How to send MavlinkV1 message through MavEsp8266.

upcfeather, 2023-11-27:
## Problem description
UDP and MavlinkProtocolV1 are required for receiving and sending Mavlink messages. Now I can normally receive Mavlink messages, but cannot send messages to the specified port or ip address. My current code is not received and sent through MavEsp8266, the specific code is as follows:

```js
import dgram from "dgram";
import {
  MavLinkPacketSplitter,
  MavLinkPacketParser,
  common,
  send,
} from "node-mavlink";
import { Duplex } from "stream";
const net = require("net");

// create udp server
const client = dgram.createSocket("udp4");
// bind mavlink service port
client.bind(8890); 

// tcp server, confirm connection
const tcpServer = net.connect({ host: "192.168.42.10", port: 8889 });
tcpServer.on("connect", () => {
  console.log("TCP Server Connected!");
});

// target port && ip
const target = { host: "192.168.42.10", port: 8890 };

// create Duplex stream
const stream = new Duplex({
  write(chunk, encoding, callback) {
    client.send(chunk, target.port, target.host, callback);
  },

  read(size) {
  },
});

const mavLinkSplitter = new MavLinkPacketSplitter();
const mavLinkParser = new MavLinkPacketParser();

client.on("message", (msg: any) => {
  mavLinkSplitter.write(msg);
});

mavLinkSplitter.pipe(mavLinkParser);

// create a registry of mappings between a message id and a data class
const REGISTRY = {
  ...common.REGISTRY,
};

mavLinkParser.on("data", (packet) => {
  const clazz = REGISTRY[packet.header.msgid];
  if (clazz) {
    const data = packet.protocol.data(packet.payload, clazz);
    // There can receive packet. 
    console.log("Received packet:", data);
  }
});

const command = new common.CommandLong();
command.targetComponent = 0;
command._param1 = 0;
command._param5 = 0;
command._param6 = 0;
command._param7 = 0;
command.confirmation = 0;
command.command = common.MavCmd.NAV_TAKEOFF_LOCAL;
// The Destination can not receive message
const result = await send(stream, command);
```
In the above code, I can normally receive messages from the target port `8890`, but I cannot send messages to the specified ip `192.168.42.10` and port `8890`. I check the documents and find that UDP should be sent through `MavEsp8266` but I find some questions after reading the source code, I hope you can help me answer them.

## Questions

1 - The sample code is the default local ip and port number, how do I set the destination ip?

When I read the source code, I found that the `MavEsp8266.start()` method accepts the `receivePort` and `sendPort` parameters, but neither parameter passes the IP, how should I pass the destination IP into it?

```js
 /**
   * Start communication with the controller via MAVESP8266
   *
   * @param receivePort port to receive messages on (default: 14550)
   * @param sendPort port to send messages to (default: 14555)
   */
  async start(receivePort: number = 14550, sendPort: number = 14555): Promise<ConnectionInfo> {
    this.sendPort = sendPort

    // Create a UDP socket
    this.socket = createSocket({ type: 'udp4', reuseAddr: true })
    this.socket.on('message', this.processIncomingUDPData)

    // Start listening on the socket
    return new Promise((resolve, reject) => {
      this.socket?.bind(receivePort, () => {
        // Wait for the first package to be returned to read the ip address
        // of the controller
        waitFor(() => this.ip !== '')
          .then(() => { resolve({ ip: this.ip, sendPort, receivePort }) })
          .catch(e => { reject(e) })
      })
    })
  }
```

2 - Using the send method of `MavEsp8266`, the built-in protocol uses `V2`, I want to send `V1` can not be sent
```js
 /**
   * Send a packet
   *
   * @param msg message to send
   * @param sysid system id
   * @param compid component id
   */
  async send(msg: MavLinkData, sysid: uint8_t = MavLinkProtocol.SYS_ID, compid: uint8_t = MavLinkProtocol.COMP_ID): Promise<number> {
    const protocol = new MavLinkProtocolV2(sysid, compid)
    const buffer = protocol.serialize(msg, this.seq++)
    this.seq &= 255
    return this.sendBuffer(buffer)
  }
```

padcom, 2023-11-28:
The current implementation works out the remote IP address just like mission-planner and qgroundcontrol does, by receiving the first packet from the target. If you really really really need to specify the IP address (e.g. if you have multiple targets) I could work something out. Currently, it is not supported.

As for sending data over [mavesp8266](https://github.com/SamJCKnox/mavesp8266), the project being referred to here only supports mavlink v2. What the protocol between the ESP8266 and the target is, is a different thing. That's why it doesn't support Mavlink V1.

If you have another bridge that does speak V1, then the proposed implementation is valid, but I'd prefer the protocol be specified as a parameter, like the `send()` function does in `mavlink.ts`. I encourage you to create a PR for it, I'll gladly review it.

Hope it helps!

padcom, 2023-11-28:
Looking at the mavesp8266 page, it says the version 1.1.1 does support v1 and 1.2.2 supports v2. I guess that answers the question if v1 should be supported.

That PR I mentioned would be very much welcomed. Please test that it works before creating the PR.

padcom, 2023-11-28:
If I may suggest, please follow the pattern of the `send()` function, where there is a default of v2, since the _latest_ version of mavesp8266 being 1.2.2 only supports mavlink v2.

upcfeather, 2023-11-28:
Thank you very much for your answer. I have already solved the problem, and I have copied your code to rewrite the `mavesp8266 `  class. Due to the special requirements, I will not create PR here, but thank you very much.

closed 2023-11-28 by upcfeather