ArduPilot/node-mavlink #31: MavEsp8266 with UDPCI endpoint (specifying target IP)

Luzzifus, 2024-01-23:
Hi,

I am currently developing a telemetry software to monitor UAV states via MavLink. Me and my team decided for various reasons to use UDPCI mode (no broadcast) on the UAV. Now our telemetry software needs to initialize sending UDP packets to the UAV via target IP, which is currently not possible with the MavEsp8266 module. 

I already wrote a local patch which allows me to set the `ip` member of a MavEsp8266 class instance through an optional parameter in the `start` function call. This works fine in our case, but I usually try to only patch dependencies locally when there's no other way. So would you be open to a pull request to add this functionality for everyone?

padcom, 2024-01-23:
@Luzzifus Create a PR, I'll review it. Please remember that the start of the conversation depends on the remote IP being known, so if you'll set it in the beginning the conversation starts pretty much immediately.

padcom, 2024-01-31:
Released with version 2.0.6

closed 2024-01-31 by padcom