ArduPilot/node-mavlink #33: MAVLink XML change to allow exponentiation operator in bitmask flag

hamishwillee, 2024-02-26:
FYI We're modifying the MAVLink XSD validation file in https://github.com/ArduPilot/pymavlink/pull/920 to allow a bitmask value to be declared using the Python exponentiation operator. 

This makes it easier to immediately visualize which bit is set by a flag, because you can use the syntax 
```xml
<entry value="2**15" name="BIT15" />
```
rather than
```xml
<entry value="32768" name=" BIT15" />
```
This is transparent to mavgen because the parser evaluates the number before creating the generated headers. However it may affect other the rust parser
We're not updating the XML yet to allow other parsers time to update.

Any concerns, please raise on https://github.com/ArduPilot/pymavlink/pull/920

padcom, 2024-02-28:
When is that going to be used in Mavlink definitions? Is it already there?

padcom, 2024-02-28:
More importantly, have you checked if it already works? The way mavlink-mappings works is it takes the content of the attribute and puts it straight in the generated code. Since Node.js already supports the `**` operator since version 7 (that's 2017 so 7 years ago) it just works with no additional code changes to the generator.

For that reason, I am hesitant to make any additional parsing/interpreting of the values in XML definitions and just rely on the language to do the right thing.

padcom, 2024-02-28:
```
/**
 * These flags encode the MAV mode.
 */
export enum MavModeFlag {
  /**
   * 0b10000000 MAV safety set to armed. Motors are enabled / running / can start. Ready to fly.
   * Additional note: this flag is to be ignore when sent in the command MAV_CMD_DO_SET_MODE and
   * MAV_CMD_COMPONENT_ARM_DISARM shall be used instead. The flag can still be used to report the armed
   * state.
   */
  'SAFETY_ARMED'                                 = 2**7,

...
```

closed 2024-02-28 by padcom

hamishwillee, 2024-02-28:
Anything that uses mavgen will work because the python parser evaluates the Python evaluation operator before generating code. So yes, it works.

We haven't rolled it out yet, but it is tested for mavgen.

Generators that use some other language for the parsing will need to make sure the ** is evaluated as a exponentiation operator. 

Probably we'll look at making some bitmap changes in a few week.s

padcom, 2024-02-28:
Ping me when you're through. I'll check it.