ArduPilot/node-mavlink #36: What's the difference between mavlink-mapping and `mavgen --lang=typescript`?

ben-xD, 2024-07-23:
Mavgen can generate typescript which imports some code from `node-mavlink` for serialization. This can be generated with `python3 mavlink/pymavlink/tools/mavgen.py --lang=TypeScript --wire-protocol=2.0 mavlink/message_definitions/v1.0/common.xml`

mavlink-mapping also has code generated from mavlink xml files, using mavlink-mapping-gen: `npx mavlink-mappings-gen@latest minimal.xml > minimal.ts`.

I wonder if you have any opinions on the differences, and which library you'd recommend using? I would definitely prefer this one over the better documentation.

Could you also share how/why ardupilot mavgen uses node-mavlink, which seems like it _competes_ with this library?

PS: This library seems really cool. I have learnt more about UDP because of this!

Thank you

padcom, 2024-07-23:
Thank you for giving the node-mavlink library a shot!

So here's the thing. The mavgen tool is used to generate fully compatible 1-to-1 library that allows you to do everything the Mavlink protocol has to offer. It is the official way of doing things. But... it comes with some drawbacks that were not acceptable for me at the time:
- no npm packages published
- C-like naming convention, which is completely different from what is being used in TypeScript
- acutally very weak type support
- no in-line documentation
- the generated code is, well... unreadable.

So at the time I thought to myself that I can do better and got to work. As a result there is a published version of the library that you don't need to generate from XML sources (although you can if you really want to, just like you showed above). All you need to do is to use npm to install the main dependency and a transport (serial or udp, which is acutally included in node.js) and you're good to go. No code generation required. Also, the published code is generated with great care and to look like it'd have been hand-crafted.

Also, the philosophy about node-mavlink is quite different. It is designed to be extensible by default as it is based on the concept of streams. That being said, the data can be piped from one stream to another, much like you pipe the standard output from one command to another in shell, for example

```bash
ls -la | sort | grep something | wc
```

To see it in action you might want to check supporting libraries such as [heartbeat](https://github.com/padcom/node-mavlink-heartbeat), [debugger](https://github.com/padcom/node-mavlink-debugger) or [ftp](https://github.com/padcom/node-mavlink-ftp) which are implemented in a very different way than what mavgen does.

The reason why this library is under the ArduPilot umbrella is also quite interesting. One day the head of this project called me and said that because they tried it out they found a bug they didn't even knew existed.

There are drawbacks to using node-mavlink. It is not a drop-in replacement nor it was never designed to be that. It is not complete (for example the parameter passing to `ParamValueTypes` (that's for reading/writing configuration options) isn't implemented yet and I don't have a good idea how to go about it. Maybe one day. Also the FTP protocol implementation is not as full as the one generated by mavgen, e.g. there is no file upload, just listing directories and downloading files. It was done to make it faster to read the settings off of the controller.

To bottom line it, it's an attempt to make the use of Mavlink easier, with in-line documentation and much less hassle than the mavgen requires. You might say it's way more TypeScripty and JavaScripty whereas mavgen is more C-like. It isn't like you can't work with the mavgen mappings, but it is going to be a much less enjoyable experience than working with node-mavlink. At least that was the reason why I did it.

closed 2024-07-23 by padcom

padcom, 2024-07-23:
Oh, and btw, mavgen's generated code seems like it's using the same library, but it isn't. It's called the same, but it is not published and the only way you can get it is to generate the code using mavgen. I do get the confusion, sorry about that. My wish was that once someone starts working with mavlink in node.js they should be able to just get it off-the-shelf rather than to play with python and mavgen and since it acutally is a node.js library for mavlink what better name to give it than.. node-mavlink :)

ben-xD, 2024-07-23:
That's really interesting and promising.

## Thinking about the limitations you mentioned

> It is not complete (for example the parameter passing to ParamValueTypes (that's for reading/writing configuration options) isn't implemented yet and I don't have a good idea how to go about it

On top of pymavlink, I implemented ftp (burst download, download, upload, list, delete, etc.) and parameter read/write. Is that what you mean by `ParamValueTypes` (I couldn't find that symbol anywhere). FTP was a bit of a pain (the CRC32 check, i was uploading corrupted messages initially), but param stuff was relatively simple:

I download all parameters when the vehicle first connects to cache them, so we can then use those properties to create a [`PARAM_SET` message](https://mavlink.io/en/messages/common.html#PARAM_SET) (we need to know the param type, `PARAM_SET.param_type`) when the user wants to change it. 

> It was done to make it faster to read the settings off of the controller.

Do you mean you use FTP to download the `.parm` file instead of downloading all params 1 by 1, slowly? That's why you didn't implement the other FTP functionality? Yup, in that case it makes sense!

I would like to contribute but because of the licensing (GPLv3), I cannot install the library in an application without needing to making the entire application GPLv3 as well. So my contributions would not be usable for some projects. Therefore I am thinking of learning from this project but building an MIT/Apache licensed one - I would rather do that than write a closed source mavlink SDK. WDYT?

padcom, 2024-07-24:
I think you've misread the licensing model here. It is not GPL - it is LGPL. I know it is more restrictive in the area of linking with other code:
[image]
All you have to do in order to comply with the license is to allow linking with another library that has the same interface. Otherwise all changes to the lib need to be given back to the original authors, which is how LGPL is actual like GPL. And I think that's perfectly fair. Also, remember that we're not talking here about the browser where all code is bundled together. It can be left as packages therefore allowing to use a different library if the end user chooses to do so.

And as for what I think about it... I think there is always room for yet another node mavlink library. I isn't an easy undertaking but if you feel brave - go for it. My only suggestion would be to first try and build it on top of node-mavlink, maybe drawing some inspiration from node-mavlink-ftp, and if that will not be good enough for you, then realize what the limitations are and start building your library with the goal of overcoming those limitations.

padcom, 2024-07-24:
[image]

ben-xD, 2024-07-25:
Thank you! I definitely misread that. I am a lot more confident now about contributing. And sure, building ontop of node-mavlink is ideal!