ArduPilot/node-mavlink #37: Can not get destination port as UDP server

JoshuaW13, 2024-09-16:
If I am running the library from a program as a UDP server I need to get the port I am receiving data from the SIM so that I know where to send commands back to. With the current implementation of the library it seems like you must know and set the send/receive ports when you start the port which restricts me to using a configuration where the sim acts as a server and my program a client. Is there any workaround to this?

padcom, 2024-09-16:
First of all, this should be a discussion, not an issue. It is a specific situation that you're experiencing where the library works as expected. Keep that in mind next time.

As for your problem please note both ports need to be specified in the start method:

```typescript
async start(receivePort: number = 14550, sendPort: number = 14555, ip: string = ''): Promise<ConnectionInfo> { ... }
```

You can also specify the IP address of the other end if you know it in advance. Otherwise it will be taken from the first received packet.

Furthermore please note the class isn't called UDP-something, it is `MavEsp6266` which points to use with the ESP8266-based serial passthrough. It might work for you as well if you can specify both ports during the call to `start(...)` but if not I suggest you make a copy of that class and tune it to your needs.

Hope it helps.

padcom, 2024-09-16:
For reference, this is the firmware this class is designed to work with:

https://github.com/BeyondRobotix/mavesp8266