ArduPilot/node-mavlink #41: Problme with MANUAL_CONTROL

KilianFa, 2025-01-06:
Hi, I'm trying to use node-mavlink to send a manual control command to pilot a rover running ArduRover.

Here’s my code:

`async function manualControl() {
    const systemId = 1; 

    const command = new common.ManualControl();
    command.target = systemId
    command.x = x
    command.y= x
    command.enabledExtensions = 0
    command.buttons = 0
    command.s = x //
    command.t = x //
    command.r = x //
    command.buttons2 = 0

    command.z = x 
    command.aux1 = 0
    command.aux2 = 0
    command.aux3 = 0
    command.aux4 = 0
    command.aux5 = 0
    command.aux6 = 0
    console.log(x)
    await send(port, command);
}`
But nothing happens on the drone. Any ideas?

padcom, 2025-01-09:
There are so many reasons why it doesn't work that without having access to the system I won't even begin to speculate. 

It might be a configuration issue, it may be connectivity issue (are you connecting via serial port or maybe via ESP8266 mavlink bridge?) then it might be a mechanical issue like disconnected servo. Just saying what you sent and what doesn't happen leaves everything else that might go wrong out making it literally impossible to diagonse.

Every system is different. Me personally, I have no experience with ArduRover, nor have a model I could convert and try to reproduce the case. I think you'll be better off trying the discussion group - there are people with much more experience with those things.

I'm closing this issue as not reproducible, but I employ you to add a comment once you figure out what the problem was - for posterity.

closed 2025-01-09 by padcom

padcom, 2025-02-10:
I don't know what more I can tell you beyond what I already explained above. You're showing me the command you try to send, then there's a gap as big as the pacific ocean and then you tell me something on the other end doesn't move. 

I'd love to help you, but as stated in the previous answer, I know nothing about your setup, means of communication or even ArduRover version. I also don't have a car I could mount ArduRover onto. You're much better off asking in discord. Maybe there is someone who already has found a solution or can direct you towards a possible one.

quocbaoevizi, 2025-02-10:
> I don't know what more I can tell you beyond what I already explained above. You're showing me the command you try to send, then there's a gap as big as the pacific ocean and then you tell me [...]

Sorry about that, I'll provide information on my case immediately.

I have a project to connect to the ardupilot to do some tasks. I already connected to a simulator ardupilot through TCP method with port and host. And already implemented some tasks like takeoff, flying to a position, and flying around using the method send messages MavLinkProtocolV2 (ex: connection.write(buffer)). Everything working perfectly.

Now I'm implementing a feature autopilot (looks like virtual joystick of QGroundControl). I'm using ManualControl API and sending messages with the old method used above but nothing happened (using QGroundControl connected to same simulator and same location still working).

I doubt the issue may be due to sysid not mapping between simulator ardupilot and my app.

padcom, 2025-02-10:
OK, now I know a bit more about the system you're trying to debug. If you say QGC works you could try out a packet sniffer and see the difference in mavlink packets being sent from QGC and the node-mavlink library. If you provide a dump of both messages maybe I'll be able to help you out figuring out what's wrong with the message you're sending.

padcom, 2025-02-10:
Essentially try the virtual joystick in QGC and see what's being sent and then try the same using node-mavlink, see what's being sent and compare the results to see what's different.

quocbaoevizi, 2025-02-11:
> OK, now I know a bit more about the system you're trying to debug. If you say QGC works you could try out a packet sniffer and see the difference in mavlink packets being sent from QGC and the [...]

Thanks for your suggestion, after putting some log debug into ardupilot source code. Finally, I figured out the reason why my code is not working. In ardupilot source code, I saw they blocked msg.sysid only working with their sysid (255), I don't know why. But now I'm hard code sysid in my code to 255 my demo feature work like a charm