ArduPilot/node-mavlink #46: How to pipe node-mavlink-ftp to MavEsp8266?

oguzmeg, 2025-04-03:
To use MavFTP with [node-mavlink-ftp](https://www.npmjs.com/package/node-mavlink-ftp) package we can use `SerialPort` as shown in the [example](https://github.com/padcom/node-mavlink-ftp/blob/master/examples/simple.ts). 

But how to use `MavEsp8266` with this package? Because `MavEsp8266` does not have a `pipe` method.

```
if (this.connectionType === 'Serial') {
      this.port = new SerialPort(options)
      const ftp = new MavFTP(this.port)
      this.reader = this.port
        .pipe(new MavLinkPacketSplitter())
        .pipe(new MavLinkPacketParser())
        .pipe(ftp)
        .resume()
      this.connected = true
    } else if (this.connectionType === 'Network') {
      this.port = new MavEsp8266() // no pipe method
      this.reader = this.port
      this.connected = true
```

padcom, 2025-04-04:
You can create a stream from MavLinkESP8266 and use it that way. Alternatively, which I didn't check, you could try using the private field `input` as it is a stream and does have the `pipe()` method.

oguzmeg, 2025-04-05:
@padcom but I can't pass MavEsp8266 to MavFTP's constructor as an argument. So how to deal with it?

padcom, 2025-04-05:
> You can create a stream from MavLinkESP8266

If you don't know how to do that I suggest you read the node.js documentation on streams. That'd be an interesting lecture.

oguzmeg, 2025-04-10:
I've created a stream from `MavEsp8266`, and it works exactly as expected. However, when I create a `MavFTP` object from that stream and pipe it back into the same stream, it works up until `ftp.downloadFile`, where the process gets stuck without throwing any errors.

`MavEsp8266` already includes a packet splitter and parser. Creating the FTP object using those existing parsers causes it to hang. I attempted some workarounds by modifying the `MavEsp8266 class`, but unfortunately, I couldn't get it to work.

```js
    if (this.connectionType === 'Serial') {
      this.port = new SerialPort(options)
      this.ftp = new MavFTP(this.port)
      this.reader = this.port
        .pipe(new MavLinkPacketSplitter())
        .pipe(new MavLinkPacketParser())
        .pipe(this.ftp)
        .resume()

      this.connected = true
    } else if (this.connectionType === 'Network') {
      const mavEsp = new MavEsp8266()
      this.port = mavEsp

      const stream = new MavLinkDuplexStream(mavEsp)

      // this.ftp = new MavFTP(stream)
      // this.reader = stream.pipe(this.ftp).resume()

      this.reader = stream

      this.connected = true
    }
```

`MavLinkDuplexStream` class definition:
```js
import { Duplex } from 'stream'
import { MavEsp8266, MavLinkData } from 'node-mavlink'

export class MavLinkDuplexStream extends Duplex {
  constructor(private mav: MavEsp8266) {
    super({ objectMode: true })

    this.mav.on('data', (packet) => {
      try {
        this.push(packet)
      } catch (err) {
        console.error('Failed to process packet:', err)
      }
    })

    this.mav.on('end', () => {
      this.push(null)
    })

    this.mav.on('error', (err: Error) => {
      console.error('MavEsp8266 error:', err)
      this.destroy(err)
    })
  }

  _read(_size: number): void {
    //
  }

  _write(
    chunk: Buffer | MavLinkData,
    _encoding: BufferEncoding,
    callback: (error?: Error | null) => void
  ): void {
    try {
      if (Buffer.isBuffer(chunk)) {
        this.mav
          .sendBuffer(chunk)
          .then(() => callback())
          .catch((err) => {
            console.error('Error sending buffer:', err)
            callback(err)
          })
      } else {
        this.mav
          .send(chunk)
          .then(() => callback())
          .catch((err) => {
            console.error('Error sending packet:', err)
            callback(err)
          })
      }
    } catch (err) {
      console.error('Exception in _write:', err)
      callback(err as Error)
    }
  }

  _final(callback: (error?: Error | null) => void): void {
    this.mav
      .close()
      .then(() => callback())
      .catch((err) => {
        console.error('Error closing connection:', err)
        callback(err)
      })
  }
}
```

oguzmeg, 2025-04-29:
This class acts like the SerialPort class. Problem solved.

```js
import { Duplex } from 'stream'
import { MavEsp8266 } from 'node-mavlink'

export class MavEspPort extends Duplex {
  private mavesp: MavEsp8266
  private input: NodeJS.ReadableStream
  public isOpen: boolean = false

  constructor(mavesp: MavEsp8266) {
    super()
    this.mavesp = mavesp
    // eslint-disable-next-line @typescript-eslint/no-explicit-any
    this.input = (mavesp as any)['input']

    this.input.on('data', (chunk: Buffer) => this.push(chunk))
    this.input.on('end', () => this.push(null))
    this.input.on('error', (err) => this.emit('error', err))

    this.once('close', () => {
      this.isOpen = false
    })

    this.isOpen = true
  }

  _read(_size: number) {
    // Data is pushed via event
  }

  // eslint-disable-next-line @typescript-eslint/no-explicit-any
  _write(chunk: any, _encoding: string, callback: (error?: Error | null) => void) {
    this.mavesp
      .sendBuffer(chunk)
      .then(() => callback())
      .catch((err) => callback(err))
  }

  async close(callback?: (err?: Error) => void) {
    try {
      await this.mavesp.close()
      this.push(null) 
      this.emit('close')
      if (callback) callback()
      // eslint-disable-next-line @typescript-eslint/no-explicit-any
    } catch (err: any) {
      this.emit('error', err)
      if (callback) callback(err)
    }
  }
}
```

closed 2025-04-29 by oguzmeg

padcom, 2025-04-29:
Nice!